EmenstaNougat/ESP32-BlueJammer #2: HSPI nRF24 module could NOT be initialized! error after installation

Max24341, 2024-08-01:
i soldered everything according to the pinout. double checked everything but it doesnt work.
the YSPI part works, but the HSPI doesnt.
im using a ESP32 Wroom 32D

this is the boot:

E (152) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
!!! HP (high power) start failure! HSPI nRF24 module could NOT be initialized! ~~~~~~~~~~~~~~~~~~~~
╦  ╦╔═╗╔═╗╦
╚╗╔╝╚═╗╠═╝║
 ╚╝ ╚═╝╩  ╩

SP Started !!! VSPI nRF24 module initialized successfully! ~~~~~~~~~~~~~~~~~~~~
SPI Frequency		= 16 Mhz
Channel			= 76 (~ 2476 MHz)
Model			= nRF24L01+
RF Data Rate		= 2 MBPS
RF Power Amplifier	= PA_MAX
RF Low Noise Amplifier	= Enabled
CRC Length		= Disabled
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 250 microseconds
Auto Retry Attempts	= 0 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Disabled
Auto Acknowledgment	= Disabled
Primary Mode		= TX
TX address		= 0xffffffffff
pipe 0 ( open ) bound	= 0xe7e7e7e7e7
pipe 1 ( open ) bound	= 0xc2c2c2c2c2
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6

EmenstaNougat, 2024-08-01:
Obviously your HSPI is not wired correctly physically or it's defect. Check everything, that's not an firmware issue.

Max24341, 2024-08-01:
> Obviously your HSPI is not wired correctly physically or it's defect. Check everything, that's not an firmware issue.

i had double checked everything, when i switch it the modules around it works. but it seems that just the HSPI is just not working

EmenstaNougat, 2024-08-01:
Then you should check another nrf on it's position

closed 2024-08-02 by EmenstaNougat