EmenstaNougat/ESP32-BlueJammer #4: Improve range

BannanaWithZipper, 2024-08-14:
Hello,

Has anybody experimented with improving the range? I am getting lost while searching for an antenna that has more range that the standard antennas that come with the nRF24L01+PA+LNA.
 Do I need more dbi? a bigger antenna? Would the antennas from a laptop be  of any good?
 A link to a specific product, ex amazon or even better AliExpress would be highly appreciated. 
What I found and thought of buying is "SAVIO Omnidirectional WiFi 12dBI 41CM RP-SMA AK-05"
https://www.savio.net.pl/en/product/savio-omnidirectional-wifi-antenna-12dbi-41cm-rp-sma-ak-05/

webportpl, 2024-08-30:
https://www.nordicsemi.com/Products/nRF24-series

EmenstaNougat, 2024-08-30:
You need a higher dbi value, but it's a directional antenna, therefore you must target the device you want to jam

closed 2024-08-30 by EmenstaNougat