EmenstaNougat/ESP32-BlueJammer #8: 10uf capacitors

cabralcr, 2024-09-30:
Hello, first of all I would like to thank you for the project, one question, what is the function of the two 10uf capacitors, wouldn't it be better to put just one of 22uf? since they are on the same 3.3v and gnd line.

Asmodeux2, 2024-10-08:
> Здравствуйте, прежде всего я хотел бы поблагодарить вас за проект, один вопрос, какова функция двух конденсаторов по 10 мкФ, не лучше ли поставить один на 22 мкФ? Так как они находятся на одной линии 3,3 В и земли.

Конденсатор нужен для стабильного запуска nrf24l01+pa+lna. Есть большое количество данных что напряжение просидает и он переходит в режим PD [Power Down] и а так конденсатор выступает в качестве фильтра.

Asmodeux2, 2024-10-08:
> Здравствуйте, прежде всего я хотел бы поблагодарить вас за проект, один вопрос, какова функция двух конденсаторов по 10 мкФ, не лучше ли поставить один на 22 мкФ? Так как они находятся на одной линии 3,3 В и земли.

В случае использования 22 мкФ нужно учесть что напряжение может пойти на один модуль а не разделиться и направится в два модуля. И от этого они умрут. [Напряжение выше 3.6V уже критично для nrf24l01]

closed 2024-10-13 by EmenstaNougat