EmenstaNougat/ESP32-BlueJammer #21: HELP

3440715446, 2024-12-26:
Why is my ESP32 indicating module installation failure？
[image]

Le0nyx, 2024-12-26:
Could you give further details, like from where you get the error, when the error happens and such. It could be a driver issue if you are running windows or based on you saying installation issue it could be when you upload that the sketch won't upload since the esp refuses connection. For this press the boot button until it is uploading with the % on the webflasher or the loadingbar in Arduino IDE with the ... is past that.

3440715446, 2024-12-28:
This is the video I uploaded, please help me identify it

3440715446, 2024-12-28:
Following your instructions, I held down the boot button and released it when prompted with%. Then, I checked the logs and found that the module installation failed. I think it's because my ESP module model is incorrect. Below is the log of the error I reported during installation
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/18265060/esp-web-tools-logs.txt)

3440715446, 2024-12-28:
> Could you give further details, like from where you get the error, when the error happens and such. It could be a driver issue if you are running windows or based on you saying installation issue it [...]

Following your instructions, I held down the boot button and released it when prompted with%. Then, I checked the logs and found that the module installation failed. I think it was because my ESP module model was incorrect. Below is the log of the error I reported during installation. The video has been accelerated

Uploading _屏幕录制20241228132857.mp4…

3440715446, 2024-12-28:
Uploading _屏幕录制20241228132857.mp4…

3440715446, 2024-12-28:
[image]

Le0nyx, 2024-12-28:
I have a similar ESP32 module so the chip itself should not be the issue. Sadly I am not able to see the video you seem to have uploaded for some reason (Might be an issue from side of Github not supporting videos). Do you have the drivers installed so the ESP32 actually shows up? If not let's try to upload a simple Sketch from Arduino IDE to the ESP. Download the ESP32 library in Arduino IDE and check the COM-Ports if the ESP shows up. If not install the necessary drivers. If it does show up try uploading a Sketch and press while the terminal says connecting the boot button on the ESP. To choose which model search for "ESP32 Dev Module". With this Sketch the built in LED on the ESP32 should blink (Define the correct Pin depending on the ESP you own):

---

// Define the onboard LED pin
// ESP8266: Built-in LED is typically on GPIO2 (D4 on NodeMCU)
// ESP32: Built-in LED is typically on GPIO2 or GPIO13
#if defined(ESP8266)
  const int LED_PIN = 2; // Change if your module has a different LED pin
#elif defined(ESP32)
  const int LED_PIN = 2; // Commonly GPIO2 or GPIO13, adjust as needed
#else
  #error "Unsupported board"
#endif

void setup() {
  // Initialize the LED pin as an output
  pinMode(LED_PIN, OUTPUT);
}

void loop() {
  // Blink the LED
  digitalWrite(LED_PIN, HIGH); // Turn the LED on
  delay(500);                  // Wait for 500 ms
  digitalWrite(LED_PIN, LOW);  // Turn the LED off
  delay(500);                  // Wait for 500 ms
}

3440715446, 2024-12-29:
[image]
[image]
I followed your instructions and uploaded two codes with different pins to ESP32, but unfortunately, there is no sign of LED flashing in my module in reality. I don't understand where the problem lies

3440715446, 2024-12-29:
[image]
According to your suggestion, I have uploaded the code for the built-in LED light flashing through Arduino IDE. It is exciting that it is indeed flashing in practice, which proves that it is not a problem with the driver. Therefore, what is the reason why I cannot install the firmware for this bluehammer

3440715446, 2024-12-29:
[image]
My error message shows that I am not currently connected to nRF24L01+PA+LNA

Le0nyx, 2024-12-29:
Great to hear. Glad I could provide some help/ hope. Happy tinkering!