EmenstaNougat/ESP32-BlueJammer #26: RC Jamming Not Working

isamsel, 2025-01-30:
To be very clear, in my build of this project, the Bluetooth and 2.4GHz WiFi jamming capabilities are EXCELLENT. Both range and robustness far exceeded my expectations and anyone claiming it just doesn't work has clearly done something incorrectly.
However, the functionality I was most excited about was the capability to jam RC car signals. I tested this functionality with the same build that has great performance otherwise and it had absolutely no effect. I don't have any model names, but I tested this functionality with at least 4 modern RC cars/radios from my dad's ample collection.

I'm enthusiastic about this project because I love introducing this flavor of chaos into the world. Because of that (and my experience in programming and electrical circuits) I would be more than happy to help debug this portion of the project, but obviously that's not exactly possible as long as the code is a black box.

Have you had any other knows issues jamming RC/Drone devices and/or would you consider allowing some outside help debugging?

EmenstaNougat, 2025-01-30:
Very happy about your feedback! :)
You should check up on the frrquencies the RC cars etc. run on. Most of them use 27mhz, There have been uploads on TikTok demonstrating it, and it jammed the signal successfully, same for me testing it on a toy drone. Make sure it's 2.4GHz, I will have another look at it the next days :)

isamsel, 2025-01-31:
I can confirm that each model I tested on had a 2.4GHz radio; I checked them all myself.
But perhaps I can think of one question that might help narrow down the issue. Which antenna(s) should be carrying the signal intended for RC and Drone jamming? On odd thing I noticed is that I never see any output from the ESP-32 ("center") antenna, which does make me wonder if there's a problem with the RF section of the microcontroller in my build (which obviously wouldn't have anything to do with your software). 
I should probably describe my setup to explain what I mean by that. My build includes the standard components purchased on Amazon; the ESP-32 and NRF24l01s are all connected to identical WiFi antennas (with correct SMA connector polarity). It's flashed with the all-in-one version of your firmware (WiFi + BT + BTLE + RC/Drone). I don't have a scope anywhere near fast enough to measure a 2.4GHz signal, but I wanted to get some relative power output measurements to gauge the quality of some RF amps I got on Amazon, so I hooked an envelope detector up to another WiFi antenna to make a crude 2.4GHz "sensor." This allows me to see when each individual antenna is outputting signal and get a rough estimate of the power levels, and based on the output from the NRF24l01s it seems to work quite well.
I had also tried flashing the ESP-32 with the RC/Drone-only version of the firmware, and while I didn't test this build on any actual cars or drones, I did look at the signal output of the antennas with the envelope detector. I never saw signal coming from any antenna with that version of the firmware, which is why I went back to the all-in-one version.

pablopeu, 2025-02-03:
Most ESP32 with antenna connectors aren't connected to the antenna connector by default, you have to remove a 0 ohm resistor (a bridge) and put it between two pads that route to the antenna connector (or a solder blob).

isamsel, 2025-02-04:
Unfortunately, I believe you're thinking of the ESP32-CAM, or at least boards that have both a PCB antenna AND an IPEX connector. I'm using the version of the ESP32-WROOM-32U that has ONLY the IPEX connector.

pablopeu, 2025-02-04:
it was worth the shoot, hope you solve it