EmenstaNougat/ESP32-BlueJammer #31: VSPI (standard power) start failure! VSPI nRF24 module could NOT be initialized!

quochung2310, 2025-03-20:
Hi, I tried all steps follow your instruction, purchased the same modules, but only HSPI module work.

[image]

This is parts I bought:

[image]

[image]

I have checked carefully, swap between modules, bought another modules, but only HSPI worked.
I thought maybe there is something wrong with ESP32 pin, but I don't know how to check.

Any suggestion please?

EmenstaNougat, 2025-03-20:
It's a simple wiring issue, hop on over to Discord for some help

Arthy71, 2025-03-20:
DISCORD LINK NOT WORKING

EmenstaNougat, 2025-03-21:
https://discord.gg/8M6XBR9X

quochung2310, 2025-03-25:
Discord link isn't valid, anything else I could do?

[image]