EmenstaNougat/ESP32-BlueJammer #59: Won't start on battery

pdvrk, 2025-07-24:
Hi, I've assembled Elecrow board version with battery and TP4056 and have strange problem. The device will turn on and work perfectly when usb cable is connected to TP4056. I can remove usb cable from TP4056 and device will continue to work on battery. But if I try to turn it on with upper slide switch only on battery, it wont turn on for some reason. I've measured ~4V voltage on slide switch and B+ and B- ports on TP4056, double checked all soldering points, still no luck.

Does anyone have same problem maybe?

pdvrk, 2025-07-24:
The problem was in connection between esp32 and battery ( - ). I've soldered small wire between B- and GND pin on ESP32 and now it works.

closed 2025-07-24 by pdvrk

pdvrk, 2025-07-24:
I've tried now to resolder TP4056 and kept soldering needle longer on contact, same problem. I also tried to bridge ESP32 GND with TP4056 OUT-, it won't start. It only works on battery power when ESP32 GND is bridged with B- on TP4056.

Do you think that it will help if I cut PH2 connector and solder battery directly on TP4056's B+ and B- ?

EmenstaNougat, 2025-07-24:
That should do it, but it seems like the solder did not reach the bottom of
the PCBs pads.
Could you please take a multimeter and test if it flows from one side on
GND of the pcb anywhere and then both of the TP4056 negative pads?

EmenstaNougat, 2025-07-25:
> ESP32 GND - TP4056 USB-C connector housing = OK
> ESP32 GND - TP4056 OUT- = OK
> ESP32 GND - TP4056 B- = NOK
>
> Maybe I have faulty TP4056, but I tried two of them, same problem. I
> bought 5-pack USB-C version from the link you provided.

The tp4056 agent faulty, don't worry.
Both positive in and output on the TP are connected to the same trace,
you'll so if one of those touches the pad, it's fine.
But the negative ones are not, as one of yours does not bave a connection
to the other GND, please try that one out with the negative input of the
JST-PH2 battery connector.

Also, maybe we should hop over to discord for a more reliable chat?

pdvrk, 2025-07-25:
As I said, at the end I've managed to power on the board with TP4056 B- soldered to one of GND pins on ESP32 with short cable. After that I've noticed that TP4056 won't charge battery (charging led remains ON all the time, with or without battery), so I've ordered a few more TP4056 boards from different supplier and will test again when they arrive. I will let you know the results on Discord.

Thanks for your help and thanks for this great board and firmware!

EmenstaNougat, 2025-07-25:
Keep me updated! :)